PX4/PX4-user_guide #2866: Parameter Reference crashes chrome tab when loaded initially

benjinne, 2023-11-09:
When accessing the parameter_reference page from its url or from google, it seems to always initially crashe my chrome tab.
The only way I'm able to access the page is by accessing the root docs page then clicking through the menu to access it.

https://docs.px4.io/main/en/advanced_config/parameter_reference.html

The tab goes unresponsive and I have to close it

[image]

I've had this issue for a long time and finally got around to posting it. It doesn't seem to matter which version of chrome I'm running and it also happens on mobile

hamishwillee, 2023-11-09:
I'm not sure what to suggest - that page works fine for me on Chrome, Firefox and Safari. Tested also a bunch of versions via Browserstack and on my phone.

Is your computer underpowered? This is a big page.

benjinne, 2023-11-09:
No it's pretty powerful and I have a strong Internet connection. Once the page is loaded it seems to work fine until I close chrome and reopen it. It's weird that if I go through the menu to the parameter reference page it always works, however using the direct link to the page causes issues when a browser has just opened.

hamishwillee, 2023-11-09:
Thank you. I'm going to close this because I don't think it is debuggable.

This page is huge and I hope to shrink it soon - actually provide vehicle-specific versions, which each will be smaller. That may well help.

benjinne, 2023-11-09:
I find myself often searching Google for px4 parameters and using the first link then it inevitably crashes my browser. I then go through the menu to access the parameter page. I can try and provide output from the browser debugger next time it happens

hamishwillee, 2023-11-09:
OK, I'll reopen. But this is almost certainly a resource issue or it would affect many other pages.

hamishwillee, 2023-11-15:
I'm going to close this because I think it may have been fixed, or will be soon by https://github.com/PX4/PX4-Autopilot/issues/22191

This removed some spurious `>` chars in the output that browsers were smart enough to ignore, but which could have caused failures. The fixes will go live next time someone modifies a parameter.

closed 2023-11-15 by hamishwillee