PX4/PX4-user_guide #2896: Error in the image of Hige-Level Software Architecture

federicociresola, 2023-11-17:
I noticed that this image has not been updated after the v1.14 release because it still contains the reference of **FastRTPS** but
now we are using the **uXRCE-DDS**

https://github.com/PX4/PX4-user_guide/blame/e62e933def6da6655bfec5d21fba0535deacf4d3/en/concept/architecture.md#L20

hamishwillee, 2023-11-23:
Thanks @federicociresola 

@beniaminopozzan Would you be happy with a fix to the diagram like this:
[image]

The bit I am not sure about is the "CDR serialization". I mean should we be talking about ROS2 here?

beniaminopozzan, 2023-11-25:
Thank @federicociresola

beniaminopozzan, 2023-11-25:
@hamishwillee , according to https://www.eprosima.com/index.php/products-all/eprosima-micro-xrce-dds (architecture section), I would put _micro CDR_ serialization or nothing. Just "via UART/UDP" like on the MAVLink block.

We don't mention the possible targets of MAVLink, then in my opinion we shuld not talk about ROS2.

hamishwillee, 2023-11-29:
All done thanks. Fixed by #2896

PR #2947 (2023-11-29): Fix uXRCE-DDS in PX4 Architecture diagram
Fixes #2896
closed 2023-11-29 by hamishwillee