PX4/PX4-user_guide #2941: GPS Altitude in Centimeters not Meters

breuerpeter, 2023-11-28:
Analyzing the topics [`sensor_gps`](https://docs.px4.io/main/en/msg_docs/SensorGps.html) and
 `vehicle_gps_position` in a ULog with PlotJuggler I notice that the "alt" fields are given in centimeters and not in meters as stated in the message reference. I am using the Holybro H-RTK F9P GNSS Series.

breuerpeter, 2023-11-28:
After uploading the [ULog](https://logs.px4.io/plot_app?log=0bfc9a29-04f9-4877-a9ec-2a0779e65ef7) I noticed that the altitude estimate is displayed in the correct units. It seems to be a problem with PlotJuggler then.

closed 2023-11-28 by breuerpeter

junwoo091400, 2023-11-28:
Yeah it does seem like GPS driver for Uboox for example correctly puts them in meter unit (converts from mm to m):

https://github.com/PX4/PX4-GPSDrivers/blob/63990d218ec35ea965d634af6a79d3155561b743/src/ubx.cpp#L1900

And I noticed that PlotJuggler displays in cm units as well, will have to debug why that's the case.

Could you open this issue in PlotJuggler repository?

Jaeyoung-Lim, 2023-11-28:
@breuerpeter I believe your log is showing the GPS altitude in mm, which is correct. The units changed to meters in this PR, and it is identifiable under a different field name (e.g. alt vs alt_m)

Jaeyoung-Lim, 2023-11-30:
@breuerpeter Because you are running a different px4 version. The field names changed with the commit that I forwarded you to.

You can check the commit you are running and check whether it is before or after the change was made

breuerpeter, 2023-11-30:
Thank you very much for the clarification @Jaeyoung-Lim. I really appreciate the support!