PX4/PX4-user_guide #2954: Build doesn't work with the latest empy

rhymesg, 2023-11-30:
I think I met this error because I'm using a git workflow that installs dependencies (including empy) every time.

[v1.13.2]
```
[  0%] Generating uORB topic headers
Traceback (most recent call last):
  File "[[tools/px_generate_uorb_topic_files.py](http://tools/px_generate_uorb_topic_files.py)](http://tools/px_generate_uorb_topic_files.py](http://tools/px_generate_uorb_topic_files.py))", line 544, in <module>
    generate_output_from_file(
  File "[[tools/px_generate_uorb_topic_files.py](http://tools/px_generate_uorb_topic_files.py)](http://tools/px_generate_uorb_topic_files.py](http://tools/px_generate_uorb_topic_files.py))", line 184, in generate_output_from_file
    return generate_by_template(output_file, template_file, em_globals)
  File "[[tools/px_generate_uorb_topic_files.py](http://tools/px_generate_uorb_topic_files.py)](http://tools/px_generate_uorb_topic_files.py](http://tools/px_generate_uorb_topic_files.py))", line 354, in generate_by_template
    em.RAW_OPT: True, em.BUFFERED_OPT: True})
AttributeError: module 'em' has no attribute 'RAW_OPT'
```
<br>
[v1.14.0] 

```
Traceback (most recent call last):
  File "/home/myname/PX4-Autopilot/Tools/msg/px_generate_uorb_topic_files.py", line 249, in <module>
    generate_output_from_file(generate_idx, f, args.outputdir, args.package, args.templatedir, INCL_DEFAULT)
  File "/home/myname/PX4-Autopilot/Tools/msg/px_generate_uorb_topic_files.py", line 166, in generate_output_from_file
    return generate_by_template(output_file, template_file, em_globals)
  File "/home/myname/PX4-Autopilot/Tools/msg/px_generate_uorb_topic_files.py", line 182, in generate_by_template
    em.RAW_OPT: True, em.BUFFERED_OPT: True})
AttributeError: module 'em' has no attribute 'RAW_OPT'
[14/895] Generating XRCE-DDS topic bridge
FAILED: src/modules/uxrce_dds_client/dds_topics.h
```

It was because the new version (4.0.0) of empy has been released 12 hours ago:
https://pypi.org/project/empy/#history

Some better guides for version or updated python codes would be helpful for other users.
[Tools/setup/requirements.txt](https://github.com/PX4/PX4-Autopilot/blob/v1.14.0/Tools/setup/requirements.txt) has `empy>=3.3`. But it doesn't work with 4.0.0.

For me, installing the previous version worked.

`pip3 install --user empy==3.3.4`

## To reproduce

Install the lastest empy:

`pip3 install --user empy`

On my PC it says "Requirement already satisfied ~ (3.3.4)".

Install empy 4.0.0 to reproduce:

`pip3 install --user empy==4.0.0`

Build PX4:

`PX4-Autopilot$ make px4_sitl`

hamishwillee, 2023-11-30:
@julianoes @junwoo091400  I'm off camping. If you are able to look at this or post query on discord that would be great.

hamishwillee, 2023-12-07:
Thanks very much for this @rhymesg . This has now been fixed by @frede791 .

closed 2023-12-07 by hamishwillee

rhymesg, 2023-12-07:
Solved by https://github.com/PX4/PX4-Autopilot/pull/22466