PX4/PX4-user_guide #3007: Does anybody know how to contact the PX4 user guide document/website management team?

angelyimou, 2024-01-04:
We are a tail-sitter drone development team from the Chinese Academy of Sciences. Last year, we contributed our team's drone flight video and team website to the PX4 user guide site (https://docs.px4.io/main/en/frames_vtol/tailsitter.html). However, we've noticed that some of the content is out of date. We hope to update it to facilitate international cooperation in the future.

I noticed that when one of my colleagues contacted the management team earlier, it was Mr. Hamish Willee who offered help.
Can Mr. Hamish Willee or other maintenance personnel continue to assist in updating? Your support means a lot to us, thank you very much for your assistance.

In order to prevent the communication delay caused by jet lag, I've drafted the content that needs to be updated as follows:

Drone Name:
ax1800->Woshark

Our website needs to be changed to:
http://uav-cas.ac.cn/WOSHARK/
(The previous version of the team site was not very clear and did not point to tail-sitter development work, so it needs to be fixed)

Thanks a lot!

hamishwillee, 2024-01-04:
Hi Angie

Thanks - this is the right place for making suggestions and letting us know about documentation problems.

The docs are open source, so in general you [can make the changes yourself as needed](https://docs.px4.io/main/en/contribute/docs.html). I have done it for you in this case - see #3009.

Regards
Hamish

closed 2024-01-04 by hamishwillee

angelyimou, 2024-01-05:
Thank you for your kindness and teaching! We'll take care of updating the content ourselves.

hamishwillee, 2024-01-07:
Thank you very much.

hamishwillee, 2024-05-16:
Hi @angelyimou Unfortunately this link has now broken. I have replaced temporarily with http://www.laarlab.cn/#/

PR #3009 (2024-01-04): tailsitter - Woshark updates
Fixes #3007