PX4/PX4-user_guide #3028: How to use MAV_CMD_DO_SET_ACTUATOR in MATLAB?

Everglow0214, 2024-01-31:
### Settings
Pixhawk 4
PX4 1.14.0
MATLAB R2023b

### Question
I want to send rotor commands to Pixhawk from MATLAB through telemetry radio. After some search, I think `MAV_CMD_DO_SET_ACTUATOR` is the appropriate interface. I can arm Pixhawk from MATLAB using `MAV_CMD_COMPONENT_ARM_DISARM`. However, I cannot make `MAV_CMD_DO_SET_ACTUATOR` work. My code is as follows.

```MATLAB
tele_com_port = "COM16";
device = serial(tele_com_port, "BaudRate", 57600);

dialect = mavlinkdialect("common.xml");
mavlink = mavlinkio(dialect);

% Arm, MAV_CMD_COMPONENT_ARM_DISARM 
cmd_arm = createcmd(dialect, "long", 400);
cmd_arm.Payload.target_system = uint8(1);
cmd_arm.Payload.param1 = single(1);
buffer_cmd_arm = serializemsg(mavlink, cmd_arm);

% Set rotor input, MAV_CMD_DO_SET_ACTUATOR
cmd_set_actuator = createcmd(dialect, "long", 187);
cmd_set_actuator.Payload.target_system = uint8(1);
cmd_set_actuator.Payload.param1 = single(0.1);
cmd_set_actuator.Payload.param2 = single(0,1);
cmd_set_actuator.Payload.param3 = single(0,1);
cmd_set_actuator.Payload.param4 = single(0.1);
buffer_cmd_set_actuator = serializemsg(mavlink, cmd_set_actuator);

fclose(device);
fopen(device);

% Arm.
fwrite(device, buffer_cmd_arm, "uint8");
fprintf("Armed\n");
% Set rotor inputs.
for i = 0: 1000
    thr = (0.8 - 0.2) / 1000 * i + 0.2; 
    cmd_set_actuator.Payload.param1 = single(thr);
    cmd_set_actuator.Payload.param2 = single(thr);
    cmd_set_actuator.Payload.param3 = single(thr);
    cmd_set_actuator.Payload.param4 = single(thr);
    buffer_cmd_set_actuator = serializemsg(mavlink, cmd_set_actuator);
    % fwrite(device, buffer_cmd_set_actuator, "uint8");
    fwrite(device, buffer_cmd_set_actuator);
    pause(0.02);
end
fprintf("Done\n");
fclose(device);
```

I checked the document at [https://github.com/PX4/PX4-user_guide/blob/main/en/payloads/README.md#generic-actuator-control](https://github.com/PX4/PX4-user_guide/blob/main/en/payloads/README.md#generic-actuator-control). However, there is no `Peripheral via Actuator Set` in my QGC. The current settings about the rotors are as follows.
[image]
[image]

Thanks in advance.

hamishwillee, 2024-01-31:
Unfortunately this feature is not present in 1.14 - see https://github.com/PX4/PX4-Autopilot/issues/21966

It is still in the docs because it should not have been removed and I have been trying to get it added back.

bkueng, 2024-02-01:
@Everglow0214 `Offboard Actuator Set` is correct, it got renamed to `Peripheral via Actuator Set` after 1.14.
You have to set `COM_PREARM_MODE` to `Always`, if you want to use it while disarmed.
And it is working on 1.14 as well.

Everglow0214, 2024-02-01:
> @Everglow0214 `Offboard Actuator Set` is correct, it got renamed to `Peripheral via Actuator Set` after 1.14. You have to set `COM_PREARM_MODE` to `Always`, if you want to use it while disarmed. And it is working on 1.14 as well.

Thanks for your reply.

I can arm Pixhawk now. In the code above, I set the rotor inputs to increase linearly. But the rotors' speed seem does not change. And Pixhawk disarms even when the program above is still running.

I guess I have to try a lower version of PX4.

hamishwillee, 2024-02-01:
You might

closed 2024-02-01 by hamishwillee