PX4/PX4-user_guide #3094: VIO guide for px4 v1.14 and above

limshoonkit, 2024-03-10:
Hi,

Need clarity on the VIO guide
https://docs.px4.io/main/en/computer_vision/visual_inertial_odometry.html#visual-inertial-odometry-vio

For firmware version v1.14 and above, there is no longer the [configuration parameter](https://docs.px4.io/main/en/advanced_config/parameter_reference.html) for MAV_ODOM_LP, is the following still relevant?
```
Set the PX4 parameter MAV_ODOM_LP to `1``. PX4 will then stream back the received external pose as MAVLink [ODOMETRY](https://mavlink.io/en/messages/common.html#ODOMETRY)[ (opens new window)](https://mavlink.io/en/messages/common.html#ODOMETRY) messages. You can check these MAVLink messages with the QGroundControl [MAVLink Inspector](https://docs.qgroundcontrol.com/master/en/qgc-user-guide/analyze_view/mavlink_inspector.html)
```

hamishwillee, 2024-03-13:
Yes, the parameter was removed in https://github.com/PX4/PX4-Autopilot/pull/20501#issuecomment-1992927367 so the guide is no longer valid. I am querying at the link about how it might be fixed.

hamishwillee, 2024-03-27:
This is invalid. I've added a note (for now) to highlight this, until someone with expertise can help fix it. https://github.com/PX4/PX4-user_guide/pull/3138

closed 2024-03-27 by hamishwillee