PX4/PX4-user_guide #3109: Unable to download Px4 simulation with MacOS

bevs8, 2024-03-17:
I have been following the User Guide and documentation for MacOS; however, inputting code inside Terminal has failed repeatedly (such as brew --cask and a variety of package downloads). 

Apparently there is a compatibility issue with the simulation software and the MacOS system that doesn't allow me to install the application. I have tried to find another source of installation, but there doesn't seem to be any other way. 

The Px4 website does not mention any issue with MacOS compatibility, so I am assuming that this is my problem. 

Any assistance is much appreciated, thanks!

hamishwillee, 2025-01-29:
Closing as stale. My understanding is that macOS instructions now work, though I have no mac to test them.

IF this is not the case, then please repost with specific information about where in the instructions things fail, and the version of macOS/mac you are running.

hamishwillee, 2025-01-29:
PS Sorry I didn't respond at the time. Generally this kind of question should be on discussion boards first, and then we fix up the docs.

closed 2025-01-29 by hamishwillee