PX4/PX4-user_guide #3111: Create Ukrainain build of PX4 User Guide

tomkaXX, 2024-03-18:
Hello, 
Our team made a 10% translation - we would love if you could create the first release with the Ukrainian version inside. We want to give it to proofreaders to check the correct translation. 

https://crowdin.com/project/px4-user-guide/uk

hamishwillee, 2024-03-18:
I will look at this tomorrow. I may not be able to build this as part of the official release - it is slow and we're pretty resource constrained. But I should be able to create build instructions that would allow you to host this on your own github fork.

hamishwillee, 2024-03-21:
Hi @tomkaXX 

OK, I cannot do this "properly" now because adding Ukranian to the main build pushes the build toolchain beyond its capacity. What I mean by this is that with 3 countries it takes 22ish minutes to build (not including deployment). When I add Ukranian the build takes more than an hour, and the build is halted by our system.

I am not sure when I will have time to look into a proper solution, but for the moment I have created a test build, so you can at least look at this:

The changes just remove the chinese and Korean builds and add a Ukranian build - allowing this to build in about 10 mins: https://docs.px4.io/v1.test/uk/ 

(you can see the changes I made in https://github.com/PX4/PX4-user_guide/compare/v1.test?expand=1)

One option would be to host this modified build yourself on github pages.

tomkaXX, 2024-03-22:
Thank you so much for your work with building it !
We are excited to see our first release!

I have some questions about updates :
1) we read and see some words translated incorrectly ( by meaning for
example ) . How do we do fixes at this stage? Return to the original file
and correct it. or do it via GitHub?
How do we allow professional proofreaders on the platform?

2) I try to leave comments in some files but I do not find how or maybe my
account is limited - can you check it?

3) Can you share a list of problems that caused a crash and we can try to
solve it for the next one?

best,
Tamara

tomkaXX, 2024-03-23:
This file contains a string starting with "- " and probably its the reason
why Crowdin not allowed to translate it.

hamishwillee, 2024-03-24:
Hi @tomkaXX 

Very happy to help your team - it's exciting to see how much you have achieved.

I have some questions about updates :

> 1) we read and see some words translated incorrectly ( by meaning for example ) . How do we do fixes at this stage? Return to the original file and correct it. or do it via GitHub? How do we allow professional proofreaders on the platform?

Tell me who you want to be a proofreader :-). I can then give them the rights to "approve" a translation. This will then be used by preference. 

Otherwise a second translation will overwrite a first translation (unless there is already an approved version).

A proofreader can provide and approve their own translation.

> 2) I try to leave comments in some files but I do not find how or maybe my account is limited - can you check it?

In bottom right of the translator screen, comments go here:

[image]

Do you see that?

> 3) Can you share a list of problems that caused a crash and we can try to solve it for the next one?

Two things.

a) The SUMMARY.md needs can't have "empty translations". You can copy the original translation from English, but not have "nothing". That's because it breaks my sidebar parsing code. I've deleted all those.

b). When you translate, you will see these markings `<0>`  some stuff `</0>`, they are replaced with HTML. It is important to make sure that they always are exactly like that - no spaces `< 0>` or <0 />` will not be detected. The result in that you end up with a mismatched tag when the Crowdin exports the output.

This doesn't break vuepress, but I am hoping to update to vitepress, and that does break with any invalid HTML. The reason to update is because vitepress is newer and better, and MAY allow me to also build the extra translation. We will see once I finish fixing the bugs.

In most cases this was done right, but care is needed. 

I have fixed or nearly fixed things that didn't work. Don't worry about it, but if you could let the team know about a) and b) that would be great.

hamishwillee, 2024-03-24:
FYI, I managed to get this to build under vitepress: https://hamishwillee.github.io/PX4-user_guide/uk/README.html

There's a long way to go in getting this as the platform on which the guide lives, but I can already see that it is much better for hosting additional translations.

tomkaXX, 2024-03-25:
Good morning!

Thank you for creating separate hosting for our translation!
It really helps to read and see parts that require improvements. Plus get a
little bit of end user experience working with a guide.

Yesterday I fixed an empty translation in SUMMARY.MD - hope it will not
break anymore.

Can you give me ( @tomacolyada) proofreader permissions and to @Radiocab -
we want to validate translation.

Thank you!

Tamara

tomkaXX, 2024-04-02:
Hello @hamishwillee 
We did around 30% translation - can you make an official build now at github?
What's the requirements for it

tomkaXX, 2024-04-02:
> Hi @tomkaXX <https://github.com/tomkaXX>
>
> Good morning to you too. OK, so I have given you "Language coordinator
> rights" over Ukranian translations. That means you can also empower people
> as proofreaders. You will need to empower radiocab because they don't have
> that name on our system when I search.
>
> To approve translations there is a little tick button, which you can see
> here. You can also delete a translation that is completely wrong, or just
> replace it with a new one.
> image.png (view on web)
>
> Thanks for fixing the empty translation :-).

also, I installed API keys for ChatGPT translation but i do not see it
visible in project. Can you help with that?

hamishwillee, 2024-04-04:
@tomkaXX Sorry for the delay, I've spent the last two days migrating to vitepress. One of the main drivers for that is so that we can build your language too. This is now done, and you can see it here: https://docs.px4.io/main/uk/

Note that bad tags will now break the build. I fixed about 6 yesterday. It's worth having a look at any Crowdin PRs (Like this https://github.com/PX4/PX4-user_guide/pull/3158). If they have errors in their workflow build step, you can look at the output to see which file might have the error.

I don't know about ChatGPS translation. But I don't have time to look at this until next week. Can you point me to the URL/instructions you followed?

Closing, as we now have a translation. Hurray!

closed 2024-04-04 by hamishwillee