PX4/PX4-user_guide #3180: 1.9.0 Documentation unavailable

eovnu87435ds, 2024-04-11:
Hello! Today I went to check on the [1.9.0 parameter reference page](https://docs.px4.io/v1.9.0/en/advanced_config/parameter_reference.html) and I noticed that I am getting a 404 error. I tried accessing from Chrome/Firefox on 2 different external IPs with google and cloudflare dns.

After looking further, the rest of the navigation links to 1.9 no longer exists. Is this an intentional change or a bug?

The reason I was looking for this is that Freefly Systems sells drones with 1.9 on it and reference the PX4 wiki in their documentation.

hamishwillee, 2024-04-11:
Hi @eovnu87435ds 

This was intentional. I needed to reduce the size of our deployment, and it never occurred to me that people would still be using v1.9. I'd prefer not to host this forever. If I host this in "some way" can you update the links on your site? I would sort this out next week.

eovnu87435ds, 2024-04-11:
@hamishwillee this is understandable! It is a very old firmware after all.  

I don't work for Freefly Systems, but I can contact them, and bring this to their attention to see if they could potentially host it, or at the very least update dead links.

I'm not very familiar with this repo, but if you could possibly point me to the commit before the removal it would certainly help!

hamishwillee, 2024-04-11:
It's actually not in this repo at all- The content here is unchanged and v1.9.0 branch is what was built. 
What gets hosted is on https://github.com/PX4/docs.px4.io in the `main` branch. Until last week it was the `master` branch which retains all the old data.

So I could fully recover this by copying the v1.9.0 tree out of master into main. However I'm thinking of a lighter-weight approach that uses a markdown renderer for these very old builds.

hamishwillee, 2024-04-11:
Either way, if you could follow up with Freefly that would be much appreciated. Don't want to leave them (or you) in the lurch.

hamishwillee, 2024-04-17:
HOwdy, did you get any response?

eovnu87435ds, 2024-04-18:
Hello! Unfortunately my request was met with radio silence :(

hamishwillee, 2024-05-01:
No worries. I think that if they haven't noticed it themselves and aren't interested following a ping, there is a limit to how much effort we should put in.

Unless you actually need this for your development, in which case I will try and find a reasonable workaround.

closed 2025-01-29 by hamishwillee