PX4/PX4-user_guide #3183: Add documentation for VectorNav integration

lashVN, 2024-04-15:


hamishwillee, 2024-04-15:
The only thing we have is https://docs.px4.io/main/en/modules/modules_driver_ins.html#vectornav, which has a broken link to Setup/usage information: https://docs.px4.io/master/en/sensor/vectornav.html

I don't even know what a vectornav is, so perhaps you could expand on what you'd like to see. I can guess from the docs at that module that it is enabled by specifying https://docs.px4.io/main/en/advanced_config/parameter_reference.html#SENS_VN_CFG on a particular serial port - but that's it.

lashVN, 2024-04-16:
I've been working on getting a driver for the VectorNav INS over the last few months, and was going to add a page for the VectorNav driver similar to [this](https://docs.px4.io/main/en/dronecan/ark_flow.html). I'm happy to work this issue and submit the work to you for a review

hamishwillee, 2024-04-16:
Thanks @lashVN - that would be excellent. Looking forward to it.

lashVN, 2024-04-18:
Hi @hamishwillee I apologize for my procedural ignorance, I haven't contributed to the user guide before, and have fairly limited contribution to public github pages (my experience is largely with private gitlab codebases)

If you'd be willing to help, where should I go to create a branch for this issue? I don't see any evident "create branch" button, and I'm prevented by git from pushing a local branch to origin.

hamishwillee, 2024-04-18:
Hi @lashVN 

Not a problem. So the broad instructions on writing docs are here: https://docs.px4.io/main/en/contribute/docs.html

You should be able to fork the guide repo at https://github.com/PX4/PX4-user_guide and push to that - then you can create a PR for to the PX4 org version.

Does that help?

lashVN, 2024-04-18:
Hi @hamishwillee , thanks for your help. 

I have a rough cut pushed [here](https://github.com/lashVN/PX4-user_guide/tree/pr-add-documentation-for-VectorNav-integration). I'd be interested in any feedback, but in particular if there is anything that is inappropriately placed. 

I mostly took a guess at where everything belongs. The VectorNav driver is in a bit of an awkward place: it's both an external sensor (IMU+GNSS), but is more commonly used as an external INS to replace PX4's internal EKF. I don't particularly care where the pages go, whatever makes the most sense to the overall architecture.

I don't see other pages that are documenting the driver behavior in the same way (such as configuration parameters set and published topics), is that appropriate?
I didn't see a place where I can link my kconfig parameters (such as CONFIG_COMMON_INS) in the same way I link my runtime parameters (such as VN_MODE). Is there somewhere else I should be looking?
Lastly, are the couple tips I have (such as dmesg or listener commands) appropriate?

hamishwillee, 2024-04-18:
Hi @lashVN 

Can you please create a PR from your branch to PX4 org version - that will make this reviewable. As it is now it is hard for me to see the delta. I work on Weds/Thurs, so will look at this next week.

In general though don't worry about "where everything goes" or whether you have put too much. My aim for these things it to write everything that I personally would need to understand "what this is for" and "when you might want to use it" and "how do I set it up from scratch, including wiring, powering, PX4 config, device config, firmware building etc. 

The only way you can do too much is if you duplicate general config stuff - in which case I will tidy that with cross links in review.

There are no docs for kconfig params. I usually just provide a link to source code for those.

lashVN, 2024-04-18:
That sounds good, I wasn't sure procedurally if you expected me to wait until it was done to submit a PR. It's still WIP, as I have to clean up some of the language. I'll fill it out a bit more to answer some of those questions in the page

hamishwillee, 2024-04-18:
Thanks. We won't merge the PR until it is ready, so WIP is fine.

PR #3191 (2024-05-09): Add VectorNav driver documentation

Tidy up work:
- [ ] Fix link to docs  in https://docs.px4.io/main/en/modules/modules_driver_ins.html#examples

Fixes #3183
closed 2024-05-09 by hamishwillee