PX4/PX4-user_guide #3202: Troubleshooting tips for ethernet and gimbal connection

Daybreakerflint, 2024-05-02:
I have been thinkering around with a gimbal and ethernet connection

I spent about 2 days trying to solve the issue until I found in a forum post from 2022 which then solved all my issues:
https://discuss.px4.io/t/fail-to-connect-pixhawk-6x-to-qgc-through-telemetry-radio/28394/3

So I turned the MAV_x_FLOW_CTRL to "Force off" (0) and then everything started to work. 

Would be nice troubleshooting information for anyone wanting to use ethernet and gimbal. Something in the works of:
_If the connection seems not to be working set MAV_x_FLOW_CTRL to "Force off" (0).
If the device is connected via a telemetry port try switching TX and RX lines._

hamishwillee, 2024-05-02:
The link you post to is about Telemetry but nothing to do with Ethernet. So I agree it might be helpful, but it would be good to understand when and why you would ever not set MAV_x_FLOW_CTRL to "Force off" (0). 

Otherwise you might as well add that comment to every single bit of information in the the docs about setting up a telemetry connection, and that isn't wise.

Let me ask a question on the boards

Daybreakerflint, 2024-05-02:
OK!
I just hope this information will at one point be able to help a few people, as I just spent two work days trying to find the issue with my setup.