PX4/PX4-user_guide #3258: The angle of the camera image does not change with the change of yaw&pitch angle during multi plane simulation

Yingqiao1998, 2024-06-19:
I just learning how to simulate plane in gz. I used `Tools/simulation/gazebo-classic/sitl_multiple_run.sh -w sonoma_raceway -s "typhoon_h480:1,rover:1" `for  simulation. And for this problem I get some tips may usful:
[Wrn] [gazebo_mavlink_interface.cpp:152] No identifier on joint. Using 0 as default sensor ID
[Wrn] [gazebo_mavlink_interface.cpp:152] No identifier on joint. Using 0 as default sensor ID
[Msg] Using MAVLink protocol v2.0
[Wrn] [gazebo_gimbal_controller_plugin.cpp:217] joint_yaw [typhoon_h480::cgo3_vertical_arm_joint] does not exist?
[Wrn] [gazebo_gimbal_controller_plugin.cpp:270] joint_roll [typhoon_h480::cgo3_horizontal_arm_joint] does not exist?
[Wrn] [gazebo_gimbal_controller_plugin.cpp:323] joint_pitch [typhoon_h480::cgo3_camera_joint] does not exist?
[Wrn] [gazebo_gimbal_controller_plugin.cpp:366] [gazebo_gimbal_controller_plugin] Streaming gimbal mavlink stream to ip: 127.0.0.1 port: 13030
Starting gazebo client

By the way, if I using /mavros/uas_2/gimbal_control/manager/configure to set gimbal, what sys_id&com_pid of primary&secondary?

hamishwillee, 2024-06-19:
@Yingqiao1998 This is not the best place to ask specific questions on working with Gazebo GZ - this is purely about the docs. Can you raise on the discussion boards?

hamishwillee, 2024-06-20:
> By the way, if I using /mavros/uas_2/gimbal_control/manager/configure to set gimbal, what sys_id&com_pid of primary&secondary?

This is a valid docs question. If you get an answer, please post link to it here.

closed 2024-06-20 by hamishwillee

Yingqiao1998, 2024-06-20:
> @Yingqiao1998 This is not the best place to ask specific questions on working with Gazebo GZ - this is purely about the docs. Can you raise on the discussion boards?

my bad:(

Yingqiao1998, 2024-06-20:
> > By the way, if I using /mavros/uas_2/gimbal_control/manager/configure to set gimbal, what sys_id&com_pid of primary&secondary?
> 
> This is a valid docs question. If you get an answer, please post link to it here.

Havent found any link. But if another setting is right, you can watch sysId&comId in mavlink status as a port used for gimbal.