PX4/PX4-user_guide #3276: Spelling error "Possibly" instead of "Possible"

FredJones4, 2024-07-05:
[image]
It says "possibly" when it should say "possible" in "If building your own branch, it is possibly you have increased the firmware size over the 1MB limit. In this case you will need to remove any drivers/modules that you don't need from the build." at https://docs.px4.io/main/en/dev_setup/building_px4.html

FredJones4, 2024-07-05:
Also, "beyond" the 1MB limit, not "over" the 1MB limit.

hamishwillee, 2024-07-06:
Thanks, fixed by #3276 - possibly a typo, but either over or beyond are fine.

FYI, feel free to make typo fixes as pull requests - makes them even easier to review.

PR #3277 (2024-07-06): Fix possibly typo
Fixes #3276
closed 2024-07-06 by hamishwillee