PX4/PX4-user_guide #3345: error: externally-managed-environment

sandeshthapa, 2024-08-13:
OS:Ubunut 22.04. 
Python3.12. 

```shell

``` bash ./PX4-Autopilot/Tools/setup/ubuntu.sh

Installing PX4 general dependencies
Hit:1 http://us.archive.ubuntu.com/ubuntu noble InRelease
Hit:2 http://us.archive.ubuntu.com/ubuntu noble-updates InRelease
Hit:3 http://packages.ros.org/ros2/ubuntu noble InRelease
Hit:4 http://us.archive.ubuntu.com/ubuntu noble-backports InRelease
Hit:5 http://security.ubuntu.com/ubuntu noble-security InRelease
Hit:6 https://packages.microsoft.com/repos/code stable InRelease
Hit:7 https://dl.google.com/linux/chrome/deb stable InRelease
Reading package lists...
Reading package lists...
Building dependency tree...
Reading state information...
astyle is already the newest version (3.1-3build1).
build-essential is already the newest version (12.10ubuntu1).
cmake is already the newest version (3.28.3-1build7).
cppcheck is already the newest version (2.13.0-2ubuntu3).
file is already the newest version (1:5.45-3build1).
g++ is already the newest version (4:13.2.0-7ubuntu1).
gcc is already the newest version (4:13.2.0-7ubuntu1).
gdb is already the newest version (15.0.50.20240403-0ubuntu1).
git is already the newest version (1:2.43.0-1ubuntu7.1).
lcov is already the newest version (2.0-4ubuntu2).
libfuse2t64 is already the newest version (2.9.9-8.1build1).
libxml2-dev is already the newest version (2.9.14+dfsg-1.3ubuntu3).
libxml2-utils is already the newest version (2.9.14+dfsg-1.3ubuntu3).
make is already the newest version (4.3-4.1build2).
ninja-build is already the newest version (1.11.1-2).
python3 is already the newest version (3.12.3-0ubuntu1).
python3-dev is already the newest version (3.12.3-0ubuntu1).
python3-pip is already the newest version (24.0+dfsg-1ubuntu1).
python3-setuptools is already the newest version (68.1.2-2ubuntu1).
python3-wheel is already the newest version (0.42.0-2).
rsync is already the newest version (3.2.7-1ubuntu1).
shellcheck is already the newest version (0.9.0-1).
unzip is already the newest version (6.0-28ubuntu4).
zip is already the newest version (3.0-13build1).
0 upgraded, 0 newly installed, 0 to remove and 3 not upgraded.

Installing PX4 Python3 dependencies
error: externally-managed-environment

× This environment is externally managed
╰─> To install Python packages system-wide, try apt install
    python3-xyz, where xyz is the package you are trying to
    install.
    
    If you wish to install a non-Debian-packaged Python package,
    create a virtual environment using python3 -m venv path/to/venv.
    Then use path/to/venv/bin/python and path/to/venv/bin/pip. Make
    sure you have python3-full installed.
    
    If you wish to install a non-Debian packaged Python application,
    it may be easiest to use pipx install xyz, which will manage a
    virtual environment for you. Make sure you have pipx installed.
    
    See /usr/share/doc/python3.12/README.venv for more information.

note: If you believe this is a mistake, please contact your Python installation or OS distribution provider. You can override this, at the risk of breaking your Python installation or OS, by passing --break-system-packages.
hint: See PEP 668 for the detailed specification.

hamishwillee, 2024-08-14:
I just tried on a completely new Ubuntu 22.04 in a VM host. Same version of Python. This all worked and ran a simulation in SITL. I didn't get that message - at the appropriate line I saww:

```
rocessing triggers for man-db (2.10.2-1) ...
Processing triggers for libc-bin (2.35-0ubuntu3.8) ...

Installing PX4 Python3 dependencies
Collecting argcomplete
  Downloading argcomplete-3.5.0-py3-none-any.whl (43 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 43.5/43.5 KB 1.0 MB/s eta 0:00:00
Collecting cerberus
  Downloading Cerberus-1.3.5-py3-none-any.whl (30 kB)
Collecting coverage
  Downloading coverage-7.6.1-cp310-cp310-manylinux_2_5_x86_64.manylinux1_x86_64.manylinux_2_17_x86_64.manylinux2014_x86_64.whl (234 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 234.7/234.7 KB 3.9 MB/s eta 0:00:00
Collecting empy==3.3.4
  Downloading empy-3.3.4.tar.gz (62 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 62.9/62.9 KB 1.4 MB/s eta 0:00:00
  Preparing metadata (setup.py) ... done
Requirement already satisfied: future in /usr/lib/python3/dist-packages (from -r /home/ubuntu/PX4-Autopilot/Tools/setup/requirements.txt (line 5)) (0.18.2)
Collecting jinja2>=2.8
  Downloading jinja2-3.1.4-py3-none-any.whl (133 kB)

```

Is your computer or environment externally managed? We don't use a venv AFAIK, so perhaps this is a restriction on your particular computer?

Anyway this script does work in the environment it intended for. So it is not a docs issue. I suggest that you create a post in the discussion board for suggestions.

closed 2024-08-14 by hamishwillee

sandeshthapa, 2024-08-20:
I do not have any venv.

sandeshthapa, 2024-08-20:
Is there anything else you think I'm missing? I  installed 22.04 in my pc. It was working fine in 20.04 a year ago.

hamishwillee, 2024-08-20:
No, everything you need should be in the script. The only thing I see in your report that is odd is the "externally managed" thing. That shows up on a google search as a common problem - with responses like https://itsfoss.com/externally-managed-environment/

However I have never seen this before myself on 22.04. That's why I suggest you create a post on the discussion board. You could try installing pipx as that suggests first.

b-kunwar, 2024-11-16:
I am trying to install UAV Toolbox Support Package for PX4 Autopilots in MATLAB Simulink. I have got the exact same issues. (My system: Windows 11, HP, MATLAB 2024a). Have you figured out a way to resolve this issue @sandeshthapa ? Any other help would be appreciated.

sandeshthapa, 2024-12-21:
@b-kunwar The issue occurred because my Ubuntu firmware was inadvertently upgraded to version 24.04 while I was downloading updates—something I only realized later. After reinstalling version 22.04, everything works as expected. I suspect (though I may be mistaken) that the problem arose because 24.04 is not yet a stable release.
 Hope that helps.