PX4/PX4-user_guide #3421: The guidance for Flight Tasks in version 1.14 is inconsistent with the actual program in version 1.14.4.

a233a2, 2024-10-30:
Hi！I want to use the newly added flight mission in version 1.14.4 of the PX4 program. Refer to the documentation of version 1.14. But I didn't find any consistency in the documentation with the code, which prevented me from adding the code correctly to the CMakeLists. I think this is causing me to not compile correctly.    
I couldn't find the following guidance to add in the program。    
[image]
Where should I add it?

hamishwillee, 2024-10-30:
@a233a2 The [flight tasks guide](https://docs.px4.io/main/en/concept/flight_tasks.html#creating-a-flight-task) is likely out of date since it was written in PX4 v1.11

Can you please post this in the [discussion boards](https://discuss.px4.io/) and then add a link back here. That will expose this problem to more people. 

@MaEtUgR Is this something you can advise on? The FLightTaskIndex looks like it is auto-generated, so I would assume that adding to the CMakeLists is the right thing to do.

More generally, can I arrange a sanity check of this doc for at least PX4v1.15 accuracy?

a233a2, 2024-10-31:
oh ok I have currently posted a similar situation to mine in the discussion section of PX4. But the bug cannot be solved, which makes me frustrated. Why does it not work when making fmu-v2_default, but it works when starting simulation?

[MyDiscussion](https://discuss.px4.io/t/new-task-when-i-make-fmu-v2-default-broken/42204)

MaEtUgR, 2024-10-31:
@a233a2 I think I know what your problem is. The structure changed slightly and we should update the documentation. According to your screenshot you added your custom task in the right place if you want it only available on targets that are not very tight on flash memory usage (`NOT px4_constrained_flash_build`). that is here: https://github.com/PX4/PX4-Autopilot/blob/e8f8bc9af7f7536543148622a8cb80cc73ab60c9/src/modules/flight_mode_manager/CMakeLists.txt#L56-L57

`fmu-v2` is one of the constrained targets so it will show in simulation (unconstrained) but not the fmu-v2 build.

Try moving the declaration of your task in CMake to the list above that which gets build on every target:
https://github.com/PX4/PX4-Autopilot/blob/e8f8bc9af7f7536543148622a8cb80cc73ab60c9/src/modules/flight_mode_manager/CMakeLists.txt#L42-L51

a233a2, 2024-11-01:
I still haven't resolved my previous mistake, but thank you all for your answers. But I changed the compilation command to FMU-V3, and now the compilation has passed. I want to know if FMU-V3 can be burned on Pixhawk 2.4.8. This temporarily solves the problem. @MaEtUgR

hamishwillee, 2024-11-06:
I strongly agree with ^^^^. The Pixhawk 2.4.8 is more than 10 years old technology with very little flash space. The money you save in buying them is lost in the time it takes to find enough flash to use the features you need.

hamishwillee, 2024-11-06:
OK. so I have added a note about contrained boards in https://github.com/PX4/PX4-user_guide/pull/3434.

I'm closing this issue now in favour of #3435. I am hoping you will provide expert review @MaEtUgR !

closed 2024-11-06 by hamishwillee