PX4/PX4-user_guide #3455: The range of parameter "MPC_XY_VEL_ALL"

zhou1214, 2024-11-13:
# User guide description：
[link of parameters](https://docs.px4.io/main/en/advanced_config/parameter_reference.html )

MPC_XY_VEL_ALL
Overall Horizontal Velocity Limit
Comment: If set to a value greater than zero, other parameters are automatically set (such as MPC_XY_VEL_MAX or MPC_VEL_MANUAL). If set to a negative value, the existing individual parameters are used.
the range of this parameter is [-20, 20]

Based on the description, this parameter should seemingly not be set to 0. 
In the simulator, when this parameter is set to 0, the drone remains stationary for a period of time and then exhibits small movements.  

# Flight log
[image]

# Question
Should the range of this parameter be adjusted to a non-continuous range, such as **[-20, -1] ∪ [1, 20]**?

hamishwillee, 2024-11-14:
I don't know. You might want to ask on the discussion boards. I won't get to look at this for at least a week.

hamishwillee, 2024-11-20:
I'm going to close this. YOu may be right, but I don't think it matters. At what point do you want to set your max vel to 0 using a parameter. The answer is never - i.e. there are better mechanisms for controlling the actual velocity and you shouldn't be using a parameter for that.

closed 2024-11-20 by hamishwillee