PX4/PX4-user_guide #3508: 404 on docs.px4.io

martinbaer, 2024-12-17:
I'm getting a 404 on https://docs.px4.io/main/

Is this happening to anyone else?

[image]

farhangnaderi, 2024-12-17:
> I'm getting a 404 on https://docs.px4.io/main/
>
> Is this happening to anyone else? [...]

It seems to be ok now: 

[image]

martinbaer, 2024-12-17:
Ah thanks, it's back up

closed 2024-12-17 by martinbaer

hamishwillee, 2024-12-17:
Yeah sorry about that. If stopped deploying 5 days ago and got stuck publishing a build from 9 months ago. Had to reset the deployment repo to get it to all work again.