PX4/PX4-user_guide #3539: QGroundControl unable to detect/calibrate Pixhawk4 sensors

ansonyuu, 2025-01-17:
Trying to set up quadcopter with Pixhawk4 flight controller, but unable to calibrate sensors using Quadcopter

## Steps to recreate
1. Download QGroundControl (using v4.4.3, but also tried older versions)
2. Upload firmware following instructions (using PX4 Stable Release 1.15.2)
3. Upload flight structure settings (successful)
4. Vehicle Setup > Sensors > Gyroscope. Produces notification "no gyros found"

## Intuitions
- Thinking that the root problem might be that it we need to update the bootloader, but when following this instruction, missing `SYS_BTL_UPDATE` in Vehicle Setup > Parameters
- Would the right move to be to build [PX4 bootloader](https://github.com/PX4/PX4-Bootloader)
- Using the advanced mode on QGC, taking the most recent build on the mater branch, still not working

```
QGroundControl can upgrade the firmware on Pixhawk devices, SiK Radios and PX4 Flow Smart Cameras.
All QGroundControl connections to vehicles must be disconnected prior to firmware upgrade.
Please unplug your Pixhawk and/or Radio from USB.
Plug in your device via USB to start firmware upgrade.
Found device: PX4 FMU V2
Error: Get Board Info: Sync: Send Command: Invalid sync response: 0x01 0x94
Upgrade cancelled
------------------------------------------
If upgrade failed, make sure to connect directly to a powered USB port on your computer, not through a USB hub. Also make sure you are only powered via USB not battery.
```

When in MAVLINK console, when we send commands `sensors status` it gives this as a result, despite being connected to the Pixhawk4 (has light/power). Doesn't change after sending `sensors stop` or `sensors start`. 

```
I am getting this error on the MAVLINK console for the QGroundControl for my pixhawk 4. I am getting this on my console after doing 'sensors status'
We are using PX4 
sensors status
selected gyro: 0 (0)
validator: best: -1, prev best: -1, failsafe: NO (0 events)
selected accel: 0 (0)
validator: best: -1, prev best: -1, failsafe: NO (0 events)
validator: best: -1, prev best: -1, failsafe: NO (0 events)
validator: best: -1, prev best: -1, failsafe: NO (0 events)
Airspeed status:
    no data
[vehicle_acceleration] selected sensor: 0, rate: nan Hz, estimated bias: [0.0000 0.0000 0.0000]
ACC 0 EN: 1, offset: [0.000 0.000 0.000], scale: [1.000 1.000 1.000], Internal
[vehicle_angular_velocity] selected sensor: 0, rate: nan Hz , estimated bias: [0.00000 0.00000 0.00000]
GYRO 0 EN: 1, offset: [0.000 0.000 0.000], Internal
vehicle_angular_velocity: gyro filter: 10617 events, 288174us elapsed, 27.14us avg, min 26us max 41us 0.959us rms
vehicle_angular_velocity: gyro filter reset: 0 events
vehicle_angular_velocity: gyro selection changed: 0 events
[vehicle_gps_position] selected GPS: 0
```

HudZah, 2025-01-17:
I'm facing this issue as well. I also get an issue where sometimes the device found switches between PX4 and Ardu pilot Chibi OS depending on the button I click with the summary or firmware first before I plug it in.

ansonyuu, 2025-01-17:
Resolved now! Pixhawk controller was an older version that is better suited for Mission Planner instead of QGC.

closed 2025-01-17 by ansonyuu

hamishwillee, 2025-01-21:
Cool. Note that this is not the right place to post bugs or support requests against Pixhawk, PX4 or QGC. It is a place to post issues with specific documentation https://docs.px4.io/main/en/

If you have issues with the software itself you would post initially in the support channels: https://docs.px4.io/main/en/contribute/support.html#forums-and-chat