PX4/PX4-user_guide #3587: unknown/unclear version reference

robertschulman1, 2025-02-19:
In en/middleware/uorb.md, reference is made to v1.16 of the PX4 distro. When I look at the tags provided at PX4/PX4-Autopilot, the tags only go up to 1.15. There is no 1.16.
If v1.16 is referring to something other than the PX4-Autopilot tags, please make the reference clearer.

hamishwillee, 2025-02-19:
I could perhaps mark the stable version here. 

[image]

robertschulman1, 2025-02-20:
> Thanks @robertschulman1 [1]
>
> image.png (view on web) [2]
>
> The format main (PX4 v1.16+) is used to indicate that this is in
> mainline and the version that the feature is expected to be present in.
> We include this because it makes it possible for us to search for the
> string and clean this up when v1.16 is released.
>
> The instance in the text body above is incorrect so I have PRd a fix
> here: #3590 [3]. I hope that is the thing you find problematic.
>
> I don't find this confusing when the PX4 version is inside brackets. If
> you have an alternative suggestion that also makes it easy to
> consistently make this maintainable let me know.

Thanks for addressing this so quickly and providing a very useful 
response.

I now understand what the maintainers are doing, but, as a newbie, I 
found the reference to v1.16+ confusing as I don't have a "filter" which 
allows me to see this reference for what it is intended to be.

Unfortunately, I do not have an alternative suggestion (unless one can 
insert meta comments with a particular pattern in the .md files as 
something to search for and potentially fix at each release).

Cheers!

bob

Links:
------
[1] https://github.com/robertschulman1
[2] 
https://github.com/user-attachments/assets/ab54ef0d-c0ae-42d1-9370-bf31f8495403
[3] https://github.com/PX4/PX4-user_guide/pull/3590
[4] 
https://github.com/PX4/PX4-user_guide/issues/3587#issuecomment-2669935846
[5] 
https://github.com/

hamishwillee, 2025-02-20:
It's a little work, but maybe we could do `main (planned for: PX4 v1.16+)`. Would that make it easier?

robertschulman1, 2025-02-20:
Indeed that would be clearer. Thanks!

Links:
------
[1] 
https://github.com/PX4/PX4-user_guide/issues/3587#issuecomment-2670292339
[2] 
https://github.com/

PR #3609 (2025-02-27): Mark PX4 v.16+ as planned
At least user found the reference to PX4 v1.16+ as confusing because there is no such release yet. This modifies that as "Planned for: PX4 v1.16" to reduce the chance of confusion.

Fixes #3587
closed 2025-02-27 by hamishwillee