PX4/PX4-user_guide #3593: CA_ROTOR{i}_CT Documentation is unhelpful and maybe incorrect

udalmia94, 2025-02-20:
In the parameter reference the documentation for the `CA_ROTOR{i}_CT` or `CA_ROTOR0_CT`, `CA_ROTOR1_CT` etc. is as follows:

> The thrust coefficient if defined as Thrust = CT * u^2, where u (with value between actuator minimum and maximum) is the output signal sent to the motor controller.

This is unhelpful for the following reasons:

1. Does not mention units - this is not a full non-dimensionalization of thrust, therefore units are important here, an example of potentially valid units here would be `N/"PWM"^2`. 
2. Is not clear what the range of `u` should be - I think I understand, but an example would go a long way here to clarify.

Also, I found evidence [in this PX4 issue report](https://github.com/PX4/PX4-Autopilot/issues/20966#issuecomment-1412326792) and [in the Advanced Tuning Guide](https://docs.px4.io/main/en/config_mc/pid_tuning_guide_multicopter.html#thrust-curve) that this description is incorrect. 

Would be super helpful if someone in the community could clarify how to correctly calculate the `CA_ROTOR{i}_CT` parameter and update the documentation at the same time.

hamishwillee, 2025-02-20:
@udalmia94 The docs are generated from the source here: https://github.com/PX4/PX4-Autopilot/blob/main/src/modules/control_allocator/module.yaml#L200-L202

The result is a normalized value between 0 and 100.

Suggest you post an issue in that repo, since that is where code issues go and cross link to here. Even better, perhaps a PR for a proposal for a fix there. If you cross link here I will see if I can get some eyes on it.

udalmia94, 2025-02-21:
> Suggest you post an issue in that repo, since that is where code issues go and cross link to here. Even better, perhaps a PR for a proposal for a fix there. If you cross link here I will see if I can get some eyes on it.

Cross post - https://github.com/PX4/PX4-Autopilot/issues/24388

Once I get a proper answer to what exactly the parameter should be, I will be happy to make a PR. 

Thank you.

udalmia94, 2025-02-21:
Sorry accidentally marked as closed. @hamishwillee can you please re-open?