PX4/PX4-user_guide #3641: Run simulation using new worlds

PBernalPolo, 2025-03-12:
I am trying and failing to run simulations using new worlds provided by OpenRobotics found in:
https://app.gazebosim.org/dashboard
A concrete example is:
https://app.gazebosim.org/OpenRobotics/fuel/worlds/Tunnel%20Simple%2002

The steps to reproduce are:

1. Clone repo with `git clone https://github.com/PX4/PX4-Autopilot.git --recursive`
2. Configure with `bash ./PX4-Autopilot/Tools/setup/ubuntu.sh`
3. Test that simulator works with
```
cd /path/to/PX4-Autopilot
PX4_GZ_WORLD=aruco make px4_sitl gz_x500
```
4. Download world, from
https://app.gazebosim.org/OpenRobotics/fuel/worlds/Tunnel%20Simple%2002
extract files, and copy them to `/path/to/PX4-Autopilot/Tools/simulation/gz/worlds`
5. Try to run simulator using
```
PX4_GZ_WORLD=simple_tunnel_02 make px4_sitl gz_x500
```
In my case I obtain the error:
`[Err] [SystemLoader.cc:92] Failed to load system plugin [dummy] : Could not find shared library.`
and after waiting for some time,
`ERROR [px4] Startup script returned with return value: 256`
I compared the file with other files defined with levels such as `/usr/share/gz/gz-sim8/worlds/tunnel.sdf` and I found that "dummy" is being defined in `tunnel.sdf` as `<plugin name="gz::sim" filename="dummy">` while in `simple_tunnel_02.sdf` it is defined as `<plugin name="ignition::gazebo" filename="dummy">`, but I don't know about the implications of this difference.

hamishwillee, 2025-03-12:
Have you read https://docs.px4.io/main/en/sim_gazebo_gz/gazebo_models.html ? The models that are supported by PX4 are in https://github.com/PX4/PX4-gazebo-models.

Next I'm going to assume that you want to know how you can bring them in. I suggest you see if you can follow the patterns in that repo and create a PR to do so. One of the team more expert than I will then explain any details/issues. If you don't get a response to that in a few days you can ping me back here and I'll see if I can find the right peope.

closed 2025-03-12 by hamishwillee

PBernalPolo, 2025-03-13:
I was focusing on:
https://docs.px4.io/main/en/sim_gazebo_gz/index.html
In particular in the section "Adding New Worlds and Models", subsection "Adding a World". There is a reference to a `CMakeLists.txt` file that should be modified to generate the targets. However, checking the git history I see that such list has been removed in 6dc39d9deb6f1de4c1bb97461193aba8ff679c59. Is it possible that the documentation is now outdated?

hamishwillee, 2025-03-13:
Thanks @PBernalPolo 

Yes, that changed last week. The author is still "firming this up" in some other PRs and wants to wait a week or two before writing docs update. I've added a specific mention of that in https://github.com/PX4/PX4-Autopilot/issues/24497#issuecomment-2722616925

dakejahl, 2025-03-13:
@PBernalPolo you likely need to edit world files you download from fuel. World plugins are now defined the server.config file (src/modules/simulation/gz_bridge). So if a world has plugins that you want to use you need to remove them from the world file and add them to the server.config. Take a look at the other world files to see what they contain.