PX4/PX4-user_guide #3649: Documentation of altitude usage (Vehicle Command)

damien-robotsix, 2025-03-19:
Hello,

This concerns https://docs.px4.io/v1.13/en/msg_docs/vehicle_command.html but might probably be also improved in other places.

The comments in vehicle_commands lacks precision concerning the altitude input.

For example, in CMD_NAV_TAKEOFF the altitude is a global altitude AMSL, ellipsoid?
I was expecting relative altitude in the first place and this is not documented.
Same in VEHICLE_CMD_SET_GPS_GLOBAL_ORIGIN
I was unable, even having a quick look in the code to know if it is using AMSL or ellipsoid reference for the altitude.

Thanks

hamishwillee, 2025-03-19:
Thanks @damien-robotsix 

Until ROS 2 came along these were not really exposed outside of the dev team, and they just copied the docs from MAVLink - i.e. this enum is a direct map to https://mavlink.io/en/messages/common.html#MAV_CMD_NAV_TAKEOFF. If we're exposing these messages through ROS, you're quite right, they need more information.

The reference for takeoff should be the home position, or more specifically, the origin of the EFK estimator on boot. So it is "alt above Home" not AMSL or ellipsoid or anything else. 

The reference for VEHICLE_CMD_SET_GPS_GLOBAL_ORIGIN is metres AMSL, noting that this is AMSL out of the GPS unit - not some standard definition of AMSL. 
This one is a bit weird because there isn't actually a command for this, it is coming from the message https://mavlink.io/en/messages/common.html#SET_GPS_GLOBAL_ORIGIN

I'll post a new issue to get some further discussion.

Some points:

1. FYI This repo is now closed for business, so I'll be closing this issue shortly. We moved all docs sources to https://github.com/PX4/PX4-Autopilot/docs last week and I'm in the process of archiving. From now on if you have issues with docs please post in the https://github.com/PX4/PX4-Autopilot repo and label with "Documentation".
2. These docs are generated from source - i.e. in main the current source is https://github.com/PX4/PX4-Autopilot/blob/main/msg/versioned/VehicleCommand.msg

hamishwillee, 2025-03-20:
FYI, confirming the rules and right way to document this in https://github.com/PX4/PX4-Autopilot/pull/24574 - closing this now.

closed 2025-03-20 by hamishwillee