PX4/PX4-user_guide #3653: https://docs.px4.io/ NOT WORKING --> We're having a really bad day.

MuhammadBilal1, 2025-03-20:
I am trying to open the documentation website, but it's not working for me since last 5-6 hours from writing this post.

Github is working absolutely fine and status also shows there is no problem on the Github end.

[image]

[image]

hamishwillee, 2025-03-20:
Yeah, I know. It's a github pages issue, out of my control. I'm attempting to revert to a build from a few days ago, which may fix it.

hamishwillee, 2025-03-20:
Working, but not latest build. Some debugging to be done

MuhammadBilal1, 2025-03-20:
@hamishwillee Thank you for the update.

hamishwillee, 2025-03-20:
Its going to go up and down today as I try work out the cause.

closed 2025-03-26 by hamishwillee