PX4/PX4-user_guide #3656: Conflicting version recommendations

claudiusli, 2025-03-31:
The documentation under the MAVSDK Integration Testing https://docs.px4.io/main/en/test_and_ci/integration_testing_mavsdk.html
says:

"[Gazebo Classic](https://docs.px4.io/main/en/sim_gazebo_classic/) is required, and should be installed by default."

That link to Gazebo Classic says:

WARNING

Gazebo Classic is supported with PX4 up to Ubuntu Linux 20.04. In Ubuntu 22.04 and later you must use [Gazebo](https://docs.px4.io/main/en/sim_gazebo_gz/) (which was [formerly known](https://www.openrobotics.org/blog/2022/4/6/a-new-era-for-gazebo) as "Gazebo Ignition").

hamishwillee, 2025-03-31:
Thanks.

@julianoes Do MAVSDK tests require Gazebo Classic, or can they now use Gazebo? - https://docs.px4.io/main/en/test_and_ci/integration_testing_mavsdk.html says it is required.

julianoes, 2025-04-01:
I don't know. Let's check what CI does:

https://github.com/PX4/PX4-Autopilot/blob/e35c1f430cdc10e7bdf82de94d14c5f50bb3837e/.github/workflows/sitl_tests.yml#L27

So this uses the image `px4io/px4-dev-simulation-focal:2021-09-08` which seems to be based on Ubuntu 20.04.

Until recently I still used Gazebo Classic on Ubuntu 22.04, but I heard the sources for it might now be gone.

hamishwillee, 2025-04-01:
@claudiusli Thanks, but I'm closing this as "not a defect" because both the posts you link are accurate, and they do not conflict. The fact is that the tests do require Gazebo Classic. They don't work on Gazebo (tested), and Gazebo/Gazebo classic do work on the versions of Ubuntu indicated.

I have created a PX4 defect because the fact the tests don't run on Gazebo will become a defect when we drop support for Ubuntu 20.04: https://github.com/PX4/PX4-Autopilot/issues/24651 - and at that point I would also have to update the docs.

PS, I know that some issue links go here, but we are shortly going to archive this repo. Future docs issues can be posted in the PX4-Autopilot repo, with the label "Documentation".

closed 2025-04-01 by hamishwillee